mchr3k/arduino-libs-manchester #30: It doesn't work with certain digital pins

SuhasRangaswamy, 2016-08-16:
Hi,
I'm using this library to transmit certain numbers from one arduino to another. It works perfectly with digital pin 2 and 3. I want to use other digital pin as well, but it didn't work with any other digital pins.
Please, tell me if I'm missing something.
This is the code.
[image]

HRip6, 2016-08-30:
I have the same issues, even worse. I cannot make it receive any packet by using man.transmit. However, when I used transmitArray, it is ok but I need more than uint8_t (I transfer a number of about 6xx-7xx). I tried every pin of the Arduino but no luck. I managed it from 2 Arduinos (Uno & Nano) but it not works. Then I tried to do it on 1 Arduino (Uno), still not working. I posted the code here (on 1 Arduino Uno):

```
#include "Manchester.h"

/*
By HRip6 [at] GitHub
Manchester library for Arduino retrieved [at] github.com/mchr3k/arduino-libs-manchester
*/

#define TX_PIN 11
#define RX_PIN 12
#define LED_PIN 13

uint8_t TX_DATA = 1;
uint8_t moo = 1;
uint8_t id; // an arbitrary message tag
uint8_t data; // actual message data
uint8_t send_id = 5;

void setup() 
{
  pinMode(LED_PIN, OUTPUT);  
  digitalWrite(LED_PIN, moo);
  Serial.begin(19200);
  man.setup(TX_PIN, RX_PIN, MAN_1200);
  man.beginReceive();
}

void loop() 
{
  uint16_t sendMessage = man.encodeMessage(send_id, TX_DATA);
  man.transmit((uint16_t)TX_DATA);
  TX_DATA++;
  
  if (man.receiveComplete()) {
    uint16_t m = man.getMessage();
    **Serial.println(m);**
    if(man.decodeMessage(sendMessage, id, data)) {
      **Serial.print("decoded... "); Serial.print(id); Serial.print("-"); Serial.print(data);**
      }
    Serial.println();
    man.beginReceive();
    moo = ++moo % 2;
    digitalWrite(LED_PIN, moo);
    }
  delay(1000);
}
```

And this is the result, the first 0 is from "m" variable got from the transmitter. As you can see from the second line, there is no issue with the encodeMessage and decodeMessage. I just put it there to test this function.

```
0
decoded... 5-4
0
decoded... 5-5
0
decoded... 5-6
0
decoded... 5-7
```

bulentperktas, 2016-09-11:
Because this version is not compatible for your sketch for now, i saw that the structure was changed in this version and not completed yet which encode and decode functions for now. you have to try old library version for your sketch, try this,

https://codebender.cc/library/Manchester

HRip6, 2016-09-14:
> ...you have to try old library version for your sketch, try this, https://codebender.cc/library/Manchester

You was right. I have tried that and everything work like a charm. Using my code above to test, the number is increased as it supposed to be :+1:  Thank you so much. Hope it works with ATTiny85 USB Module :pray:

bulentperktas, 2021-01-04:
> > ...you have to try old library version for your sketch, try this, https://codebender.cc/library/Manchester
> 
> You was right. I have tried that and everything work like a charm. Using my code above to test, the number is increased as it supposed to be 👍 Thank you so much. Hope it works with ATTiny85 USB Module 🙏

Sound is good, sure it will use with ATTiny microcontroller 👍🏻