mchr3k/arduino-libs-manchester #37: Not working on esp8266 esp-12e

troublegum, 2017-04-18:
I have receiver and transmitter. Transmitter is arduino. Receiver is esp-8266. I used your code example, and tried to connect the receiver and the transmitter. But they are not connected between themselves! But all working good when both devices are aruduino!

zeroprime790, 2017-11-01:
Hello,

Did you find a solution?

Br,
George

troublegum, 2017-11-22:
Hi, I did not find a solution.

Tvde1, 2018-07-13:
I'm also struggling with this. I really need it to work.

kiredewit, 2022-02-19:
I was also struggling to get this working. (I use a wemos D1 mini lite)

After a lot of trying, I get some communication when i split the baudrate. 

try:
ATTINY : 
man.setupTransmit(RX_PIN, MAN_600); 

ESP8266:
man.setupReceive(RX_PIN, MAN_300); 

I am a little bit lazy, so I didn`t use an antenna. Now I have problems with sending more then one byte. (number higher then 253)