mchr3k/arduino-libs-manchester #42: Transmit/getMessage don't work. Fixed.

jcmk, 2018-02-03:
Fixed by my changes. Don't really know how to use github though, so email/msg me for details if it isn't showing up. Essentially changed both to 8 bit as intended and changed the transmit function code to: uint8_t byteData[2] = {2, data};
  transmitArray(2, byteData);

fantuz, 2018-02-20:
...

`/home/max/Downloads/AR18/arduino-1.8.1/libraries/arduino-libs-manchester/Manchester.cpp:118:6: error: prototype for 'void Manchester::transmit(uint8_t)' does not match any in class 'Manchester'
 void Manchester::transmit(uint8_t data)
      ^
In file included from /home/max/Downloads/AR18/arduino-1.8.1/libraries/arduino-libs-manchester/Manchester.cpp:27:0:
/home/max/Downloads/AR18/arduino-1.8.1/libraries/arduino-libs-manchester/Manchester.h:143:10: error: candidate is: void Manchester::transmit(uint16_t)
     void transmit(uint16_t data); //transmit 16 bits of data
          ^
/home/max/Downloads/AR18/arduino-1.8.1/libraries/arduino-libs-manchester/Manchester.cpp:254:9: error: prototype for 'uint8_t Manchester::getMessage()' does not match any in class 'Manchester'
 uint8_t Manchester::getMessage(void)
         ^
In file included from /home/max/Downloads/AR18/arduino-1.8.1/libraries/arduino-libs-manchester/Manchester.cpp:27:0:
/home/max/Downloads/AR18/arduino-1.8.1/libraries/arduino-libs-manchester/Manchester.h:153:14: error: candidate is: uint16_t Manchester::getMessage()
     uint16_t getMessage(void);
              ^
/home/max/Downloads/AR18/arduino-1.8.1/libraries/arduino-libs-manchester/Manchester.cpp: In function 'uint8_t MANRX_GetMessage()':
/home/max/Downloads/AR18/arduino-1.8.1/libraries/arduino-libs-manchester/Manchester.cpp:473:30: error: ambiguating new declaration of 'uint8_t MANRX_GetMessage()'
 uint8_t MANRX_GetMessage(void)
                              ^
In file included from /home/max/Downloads/AR18/arduino-1.8.1/libraries/arduino-libs-manchester/Manchester.cpp:27:0:
/home/max/Downloads/AR18/arduino-1.8.1/libraries/arduino-libs-manchester/Manchester.h:186:21: note: old declaration 'uint16_t MANRX_GetMessage()'
     extern uint16_t MANRX_GetMessage(void);
                     ^
Using library arduino-libs-manchester in folder: /home/max/Downloads/AR18/arduino-1.8.1/libraries/arduino-libs-manchester (legacy)
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.`

fantuz, 2018-02-20:
latest git pull ... 
still using the 2 basic examples