mchr3k/arduino-libs-manchester #48: receiveComplete() never true

Miq1, 2018-03-16:
I am at a loss here: my receiver logic is close to that of the examples:

```
#include <Manchester.h>

#define ALERT 13
#define DATA_433 4

uint16_t packet;
uint16_t last_packet;

void setup() {
  pinMode(ALERT, OUTPUT);
  digitalWrite(ALERT, LOW);
  man.setupReceive(DATA_433, MAN_1200);
  man.beginReceiveArray(2, (uint8_t *)&packet);
  last_packet = 0x2f3a;
  Serial.begin(9600);
  Serial.println("");
}
```

```
void loop() {
  Serial.println(packet, HEX); // #1 - see below
  if (man.receiveComplete()) 
  {
    Serial.println(packet, HEX); // #2 - see below
    if(packet!=last_packet)
    {
      if(packet&0x4000) digitalWrite(ALERT, HIGH);
      last_packet = packet;
    }
    man.beginReceiveArray(2, (uint8_t *)&packet);
  }
}
```

the first Serial.println() (marked #1) in the loop bravely prints out the received packet, whereas #2 never prints anything at all. So it looks like receiveComplete() never is true.

What am I missing here?

Receiver is an Arduino Nano, if that matters.

Miq1, 2018-03-22:
Ideas, anyone?

Miq1, 2018-03-26:
Anyone here? :-)

joshuajayg, 2019-05-31:
Did you figure this out?  I may have an idea.

Miq1, 2019-05-31:
No, never. I finally switched to a solution where I was sending a trailing constant data pattern to detect the end of transmission.