mchr3k/arduino-libs-manchester #51: Recent merge breaks for all but ESP

paulhayes, 2018-11-23:
Commit 6d001778e43a4c5373cdbaec8e4eceea6c0f04bb breaks this library for attiny85, at328p and atmega32u4, and probably all non ESP boards.

I get the following errors
```
.piolibdeps/Manchester/Manchester.cpp: In member function 'void Manchester::setupTransmit(uint8_t, uint8_t)':
.piolibdeps/Manchester/Manchester.cpp:92:49: error: 'compensationFactor' was not declared in this scope
delay1 = (HALF_BIT_INTERVAL >> speedFactor) - compensationFactor;
^
.piolibdeps/Manchester/Manchester.cpp: In function 'void MANRX_StopReceive()':
.piolibdeps/Manchester/Manchester.cpp:490:24: error: 'timer' was not declared in this scope
timerDetachInterrupt(timer);
^
.piolibdeps/Manchester/Manchester.cpp:490:29: error: 'timerDetachInterrupt' was not declared in this scope
timerDetachInterrupt(timer);
^
```

paulhayes, 2018-11-30:
Hi @mchr3k @hjf. Have either of you checked the master branch on a non ESP chip since the pull request / merge. It's pretty broken.

mchr3k, 2018-11-30:
Okay, I've reverted.