mchr3k/arduino-libs-manchester #53: Can't transmitt on attiny85 @1Mhz

TurtleForGaming, 2019-06-25:
Here is my bootloader config:

> Clock: "1 Mhz (internal)"
> LTO: "Disabled"
> B.D.O Level: "B.D.O Disabled"
> Timer 1 clock "CPU"
> Save EPPROM: "EPPROM retained"

Here's my code:
```
#include <Manchester.h>

#define TX_PIN 3

uint8_t data[20] = {11, '1','2', '3', '4', '5', '6', '7', '8', '9','1','2','3','4','5','6','7','8','9'};

void setup() 
{
  pinMode(0, OUTPUT);
  digitalWrite(0, HIGH);
  man.workAround1MhzTinyCore(); //add this in order for transmitter to work with 1Mhz Attiny85/84
  man.setupTransmit(TX_PIN, MAN_300);
  digitalWrite(0, LOW);
  delay(2000);
}
 
uint8_t datalength=2;   //at least two data
// the loop function runs over and over again forever
void loop() 
{
  data[0] = datalength;
  
  digitalWrite(0, HIGH);
  man.transmitArray(datalength, data);
  delay(100);
  man.transmitArray(datalength, data);
  delay(100);
  man.transmitArray(datalength, data);
  digitalWrite(0, LOW);
  
  delay(5000);
  datalength++;
  if(datalength>18) datalength=2;
}
```

And the attiny dosnen't transmit when I use the 1Mhz however when I select 8MHz i works fine. Any idea ?

TurtleForGaming, 2019-06-25:
(Also using delay is off I need to put a delay of 5000ms to get an actual delay of 1sec)

Walexawos, 2023-02-16:
Hi,
I also have the same problem, with the same configuration and transmit speed.
I really need my attiny at 1MHZ, is there any available solution, other than the `man.workAround1MhzTinyCore();` ?