mchr3k/arduino-libs-manchester #56: Receive may not change data

cosmicfrogpt, 2020-10-04:
Hello, thank you for your code. It was very useful for some experiments with Conrad TX-01 temperature sensor although i had to make some changes)

I found that the receive code can sometimes not change the output data.

The bug is that the variable rx_curByte is incrementend correctly in the AddManBit function but only initialized on the program start.

It must be reseted when a complete receive is completed.

the following code resolves the problem (in the ISR):

...
if ((rx_sample == 1) &&
              (rx_curByte >= rx_maxBytes))
          {
            rx_mode = RX_MODE_MSG;
			
			//get ready for next receive !
			rx_curByte = 0;
			rx_manBits=0;
            
          }
          else
...

darkmetalman, 2021-04-14:
Hello! Which lines did  you changed?

cosmicfrogpt, 2021-04-16:
> Hello!  Which lines did you changed?

Hello, in my last experiments I changed this function(initialize lines):

MANRX_BeginReceiveBytes(uint8_t maxBytes, uint8_t *data)
{
  **rx_curByte = 0;//initialize
  rx_numMB=0;//initialize**
  rx_manBits=0;
  
  rx_maxBytes = maxBytes;
  rx_data = data;
  rx_mode = RX_MODE_PRE;
}